stoprocent/noble #4: Missing poweredOn event (at least in Ubuntu 24.04)

rgillan, 2024-08-21:
Pulling through from here:
https://github.com/abandonware/noble/issues/341

rgillan, 2024-08-21:
confirmed with the quick start that the event never happens:
```
var noble = require('@stoprocent/noble');

noble.on('stateChange', async (state) => {
  console.log('stateChange:' + state);
  if (state === 'poweredOn') {
    await noble.startScanningAsync();
  }
});

noble.on('scanStart', () => { console.log('scanStart'); });
noble.on('scanStop', () => { console.log('scanStop'); });

noble.on('discover', async (peripheral) => {
  await noble.stopScanningAsync();
  await peripheral.connectAsync();
  const {characteristics} = await peripheral.discoverSomeServicesAndCharacteristicsAsync(['180f'], ['2a19']);
  const batteryLevel = (await characteristics[0].readAsync())[0];

  console.log(`${peripheral.address} (${peripheral.advertisement.localName}): ${batteryLevel}%`);

  await peripheral.disconnectAsync();
  process.exit(0);
});
```

Apollon77, 2024-08-21:
Same question As in matter.js ...

Are capabolities set correctly? https://github.com/stoprocent/noble?tab=readme-ov-file#running-without-rootsudo-linux-specific ... else try if it works with sudo then it is about this.

rgillan, 2024-08-21:
@Apollon77 we have tested both with and without setcap (as non root) and even running directly as root still has the same issue.
This approach (set_cap_raw) is also extremely insecure as it gives this unsolicited access to any nodejs process in the machine, but that is a secondary issue for us at this point, we're just trying to get things functional and it appears that this is where it's failing on newer OS/node environments.

Apollon77, 2024-08-22:
Which kernel the newest Ubuntu uses? Could it be related to https://github.com/abandonware/node-bluetooth-hci-socket/issues/60#issuecomment-2197545245

stoprocent, 2024-08-26:
Hi Guys

I will have a look but this week is impossible for me. I still have to fix what @Apollon77 was mentioning above and I also plan to do this at the same time and we can test after.

stoprocent, 2024-09-22:
Ok guys finally got this finished I think. I wanted to refactor entire hci native module to move it to N-API and do some cleanups there first. Ive tested everything on ubuntu 24 and seems to be working just fine.

Changes are here: https://github.com/stoprocent/node-bluetooth-hci-socket/pull/2

Whaty I would like you @rgillan or @Apollon77 to do is to:

1. Clone brnach from here: https://github.com/stoprocent/node-bluetooth-hci-socket/pull/2
2. Change manually noble `package.json` to point to cloned directory
3. Test it

ab-kily, 2024-09-24:
@stoprocent hi! I checked and it seems it is not working on ubunti 24.04.1. My configuration is:
```shell
# cat /etc/issue
Ubuntu 24.04.1 LTS \n \l
# uname -a
Linux bagger-home 6.8.0-45-generic #45-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug 30 12:02:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
# cat package.json 
{
  "dependencies": {
    "@abandonware/noble": "^1.9.2-25",
    "@abandonware/bluetooth-hci-socket": "file:node-bluetooth-hci-socket",
    "noble": "^1.9.1"
  }
}
# cat node-bluetooth-hci-socket/.git/config|grep url
	url = https://github.com/stoprocent/node-bluetooth-hci-socket.git
# cat node-bluetooth-hci-socket/.git/HEAD
ref: refs/heads/feature/fix_set_filter
# cat test.js
const noble = require('@abandonware/noble');
noble.on('stateChange', function(state) {
    console.log(state);
	if (state === 'poweredOn' ) {
		console.log('on -> stateChange   StartScan');
		noble.startScanning([], true);
	} else {
		console.log('on -> stateChange   StopScan');
		noble.stopScanning();
	}
});
# sudo node test.js 
```
Nothing happens when I ran `sudo node test.js`.

stoprocent, 2024-09-24:
@ab-kily thats wierd ... 
I have published now https://www.npmjs.com/package/@stoprocent/bluetooth-hci-socket. Can you try setting node-bluetooth-hci-socket to `1.3.0` in `package.json` and npm install

stoprocent, 2024-09-24:
@ab-kily dont use `@abandonware` because im using different namespace. I see now why it wouldn't work

stoprocent, 2024-09-24:
Do this:
1. Clone this repo so: `https://github.com/stoprocent/noble`
2. Change `package.json` to  `"@stoprocent/bluetooth-hci-socket": "^1.3.0"`
3. Run node any example from this repo

ab-kily, 2024-09-25:
@stoprocent I can confirm it works! My configuration is: Ubuntu 24.04.1 (kernel 6.8.0).
```shell
$ cat package.json
{
  "dependencies": {
    "@stoprocent/noble": "file:noble",
    "@stoprocent/bluetooth-hci-socket": "^1.3.0"
  }
}

$ cat test.js
cat test.js 
const noble = require('@stoprocent/noble');
noble.on('stateChange', function(state) {
    console.log(state);
	if (state === 'poweredOn' ) {
		console.log('on -> stateChange   StartScan');
		noble.startScanning([], true);
	} else {
		console.log('on -> stateChange   StopScan');
		noble.stopScanning();
	}
});
noble.on('discover', async (peripheral) => {
    console.log(peripheral.address)
});

$ sudo node test.js
sudo node test.js
poweredOn
on -> stateChange   StartScan
8c:4b:14:19:af:7a
4a:e1:94:44:70:0b
68:ed:b8:c8:db:81
...
```
@stoprocent thank you very much!

stoprocent, 2024-09-25:
Perfect! I can update `noble` now :) Thank you for the verification

stoprocent, 2024-09-25:
Ok I'm closing this issue as it was released now under https://github.com/stoprocent/noble/releases/tag/v1.14.2

Apollon77, 2024-09-25:
I assume I will also come to test it next days in matter.js

marcelkottmann, 2024-09-28:
Hi @stoprocent , I had the same problem (after upgrading to ubuntu 24 several months ago) and your fix solved the issue. But after exactly 60 seconds communication with the bluetooth device the connection gets disconnected by "something". That is an issue I did not have with the previous noble version I used ( "@abandonware/noble": "1.9.2-24" ). But even in a virtualbox vm with Ubuntu 22 and the old version noble will not start anymore, so I am not able to re-test that.
Can you ensure "the disconnect after 60 secs communication" was not introduced by the fix?

grougy, 2024-09-28:
First thank you for all the work!

I confirm it is solving (and seems compatible) issue with [node-poweredup](https://github.com/nathankellenicki/node-poweredup) on Ubuntu 22.

SImple receipe:
```
  "dependencies": {
    "node-poweredup": "^9.0.0"
  },
  "overrides": {
    "@abandonware/noble": "npm:@stoprocent/noble"
  }
```
(For those you may struggle, package-lock may be removed and a fresh npm install needed. And @stoprocent version will be inside an @abandonware folder within node_modules)

I am also seeing regular disconnection, will try to give more details.

marcelkottmann, 2024-09-29:
@stoprocent @grougy I found a possible fix for the 60 seconds disconnect problem: see https://github.com/stoprocent/noble/issues/9

Apollon77, 2024-09-29:
@grougy maybe open an issue with [node-poweredup](https://github.com/nathankellenicki/node-poweredup) to switch the lib :-)

stoprocent, 2024-09-30:
I will be working on @Apollon77 issue and I can look at @marcelkottmann comments from #9 and will update you guys in #9

stoprocent, 2024-10-01:
Ok guys I'm closing this one as for me it is fixed. I will reopen if needed

closed 2024-10-01 by stoprocent