stoprocent/noble #7: Strange error when installing noble

Apollon77, 2024-09-27:
see https://github.com/Apollon77/hap-controller-node/actions/runs/11073038352/job/30768679102#step:4:13

```
npm error code 1
npm error path /home/runner/work/hap-controller-node/hap-controller-node/node_modules/@stoprocent/noble
npm error command failed
npm error command sh -c patch-package
npm error patch-package 8.0.0
npm error Applying patches...
npm error Error: Patch file found for package docker-pull which is not present at node_modules/@vweevers/docker-pull
npm error Error: Patch file found for package docker-run which is not present at node_modules/docker-run
npm error Error: Patch file found for package prebuildify-cross which is not present at node_modules/prebuildify-cross
npm error ---
npm error patch-package finished with 3 error(s).
```

That more sound like patch stuff that is only relevant for your CI pipeline and should never trigger errors when used normally, or?

Apollon77, 2024-09-27:
https://github.com/stoprocent/noble/commit/c2feb69551845ba451ebb8a4c591345550babe05#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519R67 is causing

stoprocent, 2024-09-27:
This has been fixed with `1.14.3` release

closed 2024-09-27 by stoprocent

Apollon77, 2024-09-27:
verified, thank you!