stoprocent/noble #9: Disconnect after 60 seconds

marcelkottmann, 2024-09-29:
As already commented in https://github.com/stoprocent/noble/issues/4 I experience a disconnect event 60 seconds after establishing a connection in the new implementation.

I debugged this problem and found a bug in [node-bluetooth-hci-socket](https://github.com/stoprocent/node-bluetooth-hci-socket):

The `L2_CONNECT_TIMEOUT` in `node-bluetooth-hci-socket/include/BluetoothStructs.h:36` must always be lower than the timeout in `node-bluetooth-hci-socket/lib/native.js:20`. Both are currently set to 60 seconds.

**Fix**: Reducing `L2_CONNECT_TIMEOUT` to 59 seconds fixes the problem.

Observations: Reducing both values to 30 secs lead to a disconnect after 30 seconds. Then reducing L2_CONNECT_TIMEOUT to 29 seconds then again fixes the problem.

stoprocent, 2024-10-01:
[This](https://github.com/stoprocent/noble/pull/10) should fix the issue. The problem wasn't with the time itself but with how connections were queued. It surfaced after I migrated the code to more modern APIs. Let me know if it works on your end. Eventually, I was able to reproduce and find the issue and fix it.

stoprocent, 2024-10-03:
@marcelkottmann  can you confirm ?

marcelkottmann, 2024-10-04:
Hi @stoprocent , I can confirm that the disconnects disappear after upgrading to the new version. Thank you for maintaining this package!

closed 2024-10-04 by marcelkottmann