stoprocent/noble #11: Missing "service data" on Windows

spudwebb, 2024-10-09:
It looks like on Windows the "service data" are not correctly parsed and passed back but these are required by the Matter standard. Using the same bluetooth dongle and the same Matter device, the "service data" are correctly reported.

Linux:
2024-10-08 14:48:59.885 DEBUG  NobleBleClient       Found peripheral 30:de:4b:db:7f:24 (undefined): {"localName":"undefined","txPowerLevel":"undefined","manufacturerData":"undefined",**"serviceData":[{"uuid":"fff6","data":{"type":"Buffer","data":[0,206,4,136,17,1,1,1]}}]**,"serviceUuids":["8641","fff6"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]}

Windows:
2024-10-08 15:13:19.756 DEBUG  NobleBleClient       Found peripheral 30:de:4b:db:7f:24 (): {"localName":"","txPowerLevel":0,"manufacturerData":{"type":"Buffer","data":[]},**"serviceData":[]**,"serviceUuids":["8641","fff6"]}

Reference discussion with @Apollon77 : https://github.com/project-chip/matter.js/discussions/1241

stoprocent, 2024-10-10:
What kind of BLE implementation you are using? Native windows or USB ?

spudwebb, 2024-10-10:
I'm using a TP-Link UB500 USB adapter on my desktop PC

But I saw the exact same problem using the native BLE from my windows laptop.

stoprocent, 2024-10-10:
I'm on a business trip right now. I will be back soon and I will have a look. No probs

leonardocavagnis, 2024-11-12:
Hi @stoprocent,
I'm experiencing the same issue. On macOS, service data is parsed correctly, but on Windows 10, it isn't.

@spudwebb Any updates?

spudwebb, 2024-11-12:
No update on my side, but I would love to make this work.

@stoprocent , let us know  how we can help to fix this.

stoprocent, 2024-11-12:
Hi Guys I had a crazy time at work. I will try to look at this end of the current week. Stay tuned

stoprocent, 2024-12-21:
I've just compiled noble on my windows 11 and this is what im getting:
[image]
Im getting full advertising data, maybe the problem isnt Noble?

stoprocent, 2024-12-22:
Ok. I found the issue and it's not just windows :) Issue was `noble` was/is only parsing advertising data when in scan response and not advertising frames. That was a mistake by original author. I wil fix it this week.

stoprocent, 2024-12-22:
@Apollon77 @spudwebb  @leonardocavagnis I've added this feature now for native Windows. It wasn't even the above. Windows native implementation didn't even parse the data. Now I've added the same behavior in all platforms: Linux, Mac and Windows. Test it please and let me know so I can close the issue :)

leonardocavagnis, 2024-12-23:
@stoprocent Sure! Thanks! I'll test it soon and get back to you with feedback!

Apollon77, 2024-12-25:
Thanks a lot @stoprocent. Matter.js nighly is using the latest noble version as dependency. So can be used to test.

stoprocent, 2024-12-27:
Any update @leonardocavagnis  or @Apollon77. 
BTW @Apollon77 I want to do `2.0` version soon. 
I would like to ditch all websocket-related stuff and just keep: `Native Win`, `Mac`, `Linux` and `HCI` over `UART` and `USB`. I've started converting code to TypeScript and doing many other cleanups. Do you have any more ideas for the lib?

Apollon77, 2024-12-30:
Hey @stoprocent,

I asked in the matter.js community (I myself do not use windows) ... but no feedback so far, sorry. I would expect that also the holidays are reasons and hope we get feedback when people come back in january.

TS conversion sounds awesome. Fully support it!!

I still have this issue on noble side that when device unexpected disconnects directly after connect and first command is then send that "something" does an automatic reconnect and resends the command - but the response can not be mapped and so the command "hangs" and never resolves. If you have an idea where this auto-reconnect happens then I might also be also to look ... or maybe you already fixed that one by the TS conversion or such :-)
(Also repro that is PITA because just happens sometimes ;-) )
Maybe once you have a new codebase release as "beta" or "dev" tag to npm and I can test it and provide new log.

Re other idea I will send you one thing in slack.

stoprocent, 2024-12-30:
As far as I was testing it it's completed and working just fine on windows 10 and 11. I will close the issue and if anything will be wrong with the implementation let's create a new one @spudwebb  @Apollon77  and @leonardocavagnis

closed 2024-12-30 by stoprocent

Apollon77, 2024-12-31:
I agree. I also now have another ble test device so I will try to check things next with newest version of noble. Awesome work!

spudwebb, 2025-01-08:
@stoprocent , I confirm that the problem described in my first post is fixed, the device is now correctly discovered by matter.js on Windows, Thanks for the fix.

However, I found some other problems with other devices on windows, see https://github.com/stoprocent/noble/issues/20