stoprocent/noble #13: Connect error 62

Apollon77, 2025-01-01:
Whats this error? I still get often this error when trying to connect to several devices ... also log looks a bit strange because in the middle the scanning gets enabled again .. no ideal if this is planned? in my code it is not triggered

```
matter> commission pair 33333 --pairingCode 02397732725 --ble
2025-01-02 00:23:27.446 INFO   Node                 matter.js shell controller started for node 1
2025-01-02 00:23:27.459 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: 0.0.0.0:5353
2025-01-02 00:23:27.475 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: :::5353
2025-01-02 00:23:27.481 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: 0.0.0.0:5353
2025-01-02 00:23:27.493 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: :::5353
2025-01-02 00:23:27.499 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:undefined localAddress: :::47064
2025-01-02 00:23:27.502 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:47064 localAddress: 0.0.0.0:47064
  hci reset - writing: 01030c00 +0ms
  hci setting filter to: 1600000020c10800000000400000 +2ms
  hci le read supported feature - writing: 01032000 +0ms
  hci onSocketData: 040e0401030c00 +4ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3075 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +2ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +1ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0c010320003f00000800000000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8195 +1ms
  hci 		status = 0 +0ms
  hci 		result = 3f00000800000000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +0ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +0ms
  hci write LE host supported - writing: 016d0c020100 +1ms
  hci read LE host supported - writing: 016c0c00 +0ms
  hci le read buffer size - writing: 01022000 +0ms
  hci read bd addr - writing: 01091000 +1ms
  hci onSocketData: 040e0401010c00 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3073 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0401012000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8193 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0c010110000972010931011961 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4097 +0ms
  hci 		status = 0 +0ms
  hci 		result = 0972010931011961 +0ms
  hci set scan enabled - writing: 010c20020001 +1ms
  hci set scan parameters - writing: 010b200701120012000000 +0ms
  hci onSocketData: 040e0a01091000efe53e32a6dc +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +0ms
  hci 		result = efe53e32a6dc +0ms
  hci address = dc:a6:32:3e:e5:ef +0ms
  noble addressChange dc:a6:32:3e:e5:ef +0ms
  hci onSocketData: 040e0401010c00 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3073 +0ms
  hci 		status = 0 +0ms
[... 418 lines cut ...]
  hci 		reason = 62 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
2025-01-02 00:23:29.266 ERROR  BleChannel           Peripheral 8c:bf:ea:94:99:56 disconnected while trying to connect, try again 62
2025-01-02 00:23:29.267 DEBUG  BleChannel           BLE peripheral state disconnected
2025-01-02 00:23:29.267 DEBUG  BleChannel           Connect to Peripheral now (try 3)
  hci set scan enabled - writing: 010c20020001 +2ms
  hci onSocketData: 040e04010c200c +7ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +1ms
  hci onSocketData: 0413050140000100 +230ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 1 +1ms
  hci 		closed +0ms
  hci flush - pending: 0 queue length: 0 +0ms
```

you can see here matter,js log (yes I did not had a chance to downstrip it, sorry, will try next)
* Device gets discovered
* We connect to it ... this ends in error 62 ... then another error 62 and then on third attempt it "hangs" afterthe last two logs ... connectAsync never returns.
Do you see anything?

Apollon77, 2025-01-01:
PS: in matter.js each start/stop scan commands is also logging an extraline ... which is not logged

Apollon77, 2025-01-03:
@stoprocent There is also an other issue here.

it seems that this special devcie (its it because of the public address?) is somehow starting a scan during connection.
As you can see below the "2025-01-02 00:23:27.844 DEBUG  BleChannel           Connect to Peripheral now (try 1)" line ... in this case also "startScan" event is emitted which means that my program thinks that scanning is in progress ... a stopScan is not emitted ... this means if I want to restart scanning in the same process this will fail ...
* first because my could would not start again because it thinks that scan is already active
* second ... if I do stopScanning this would never work and await will never be triggered as it seems because it already is stopped

Apollon77, 2025-04-19:
Updated log with Noble 2.0.1

```
2025-04-19 14:23:43.053 DEBUG  NobleBleClient       Found peripheral fb:71:e2:3f:8d:8b (SchuecoSensor): {"localName":"SchuecoSensor","txPowerLevel":"undefined","manufacturerData":"undefined","serviceData":[{"uuid":"fff6","data":{"type":"Buffer","data":[0,135,9,241,255,4,128,1]}}],"serviceUuids":[],"serviceSolicitationUuids":[]}
2025-04-19 14:23:43.057 DEBUG  BleScanner           Discovered device fb:71:e2:3f:8d:8b 008709f1ff048001
2025-04-19 14:23:43.059 DEBUG  BleScanner           Discovered device fb:71:e2:3f:8d:8b data: {"deviceIdentifier":"fb:71:e2:3f:8d:8b","D":2439,"SD":9,"VP":"65521+32772","CM":1,"addresses":[{"type":"ble","peripheralAddress":"fb:71:e2:3f:8d:8b"}]}
2025-04-19 14:23:43.060 DEBUG  BleScanner           Finishing waiter for query SD:9, resolving: true
2025-04-19 14:23:43.061 DEBUG  NobleBleClient       Stop BLE scanning for Matter Services ...
  hci set scan enabled - writing: 010c20020001 +12ms
  hci onSocketData: 043e2102010001bc37d160ce4f1502011a0eff4c000f059000e9fcb610022404020a00d6 +0ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 33 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 0001bc37d160ce4f1502011a0eff4c000f059000e9fcb610022404020a00d6 +0ms
  hci 			type = 0 +0ms
  hci 			address = 4f:ce:60:d1:37:bc +0ms
  hci 			address type = random +0ms
  hci 			eir = 02011a0eff4c000f059000e9fcb610022404020a00 +1ms
  hci 			rssi = -42 +0ms
  hci onSocketData: 043e0c02010401bc37d160ce4f00dc +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +1ms
  hci 		LE meta event data length = 12 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 0401bc37d160ce4f00dc +0ms
  hci 			type = 4 +0ms
  hci 			address = 4f:ce:60:d1:37:bc +0ms
  hci 			address type = random +0ms
  hci 			eir =  +0ms
  hci 			rssi = -36 +0ms
  hci onSocketData: 043e0f02010000dd84c388b6f403020118b2 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +1ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 15 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 0000dd84c388b6f403020118b2 +0ms
  hci 			type = 0 +0ms
  hci 			address = f4:b6:88:c3:84:dd +0ms
  hci 			address type = public +0ms
  hci 			eir = 020118 +0ms
  hci 			rssi = -78 +0ms
  hci onSocketData: 043e2702010301ffcb7d9a52431b02011a17ff4c0009081371c0a8b2201b58160800bd96827dd48374bc +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 39 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 0301ffcb7d9a52431b02011a17ff4c0009081371c0a8b2201b58160800bd96827dd48374bc +0ms
  hci 			type = 3 +0ms
  hci 			address = 43:52:9a:7d:cb:ff +0ms
  hci 			address type = random +0ms
  hci 			eir = 02011a17ff4c0009081371c0a8b2201b58160800bd96827dd48374 +0ms
  hci 			rssi = -68 +0ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble scanStop +228ms
2025-04-19 14:23:43.071 INFO   ControllerDiscovery  Found 1 devices using identifier {"shortDiscriminator":9}
2025-04-19 14:23:43.073 DEBUG  ControllerDiscovery  Server addresses to try: ble://fb:71:e2:3f:8d:8b
2025-04-19 14:23:43.074 DEBUG  ControllerDiscovery  Try to communicate with ble://fb:71:e2:3f:8d:8b ...
2025-04-19 14:23:43.075 INFO   PeerCommissioner     Establish PASE to device VP: 65521+32772
2025-04-19 14:23:43.078 DEBUG  BleChannel           Peripheral fb:71:e2:3f:8d:8b: Connect to Peripheral now (try 1)
  hci create le conn - writing: 010d20196000300000018b8d3fe271fb000600120000002a0004000600 +10ms
[... 87 lines cut ...]
  hci write acl data packet - writing: 024000070003000400020001 +0ms
  hci onSocketData: 011620024000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8214 +0ms
  hci 		data len = 2 +0ms
  hci onSocketData: 040f0400011620 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +1ms
  hci 		cmd = 8214 +0ms
  hci onSocketData: 043e0c043e40003f00000800000000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 4 +0ms
  hci 		LE meta event data length = 12 +0ms
  hci 		LE meta event num reports = 62 +0ms
  hci 		LE meta event data = 40003f00000800000000 +0ms
  hci onSocketData: 0405040040003e +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +0ms
  hci 		handle = 64 +0ms
  hci 		reason = 62 +0ms
  hci flush - pending: 0 queue length: 0 +1ms
2025-04-19 14:23:43.697 ERROR  BleChannel           Peripheral fb:71:e2:3f:8d:8b disconnected while trying to connect, try again 62
```

Apollon77, 2025-04-19:
Grmpf ... The problem is that this happens basically 80% of the cases ... I will try to get my retry mechanism working again after update

stoprocent, 2025-04-19:
Was it not there before? 
1. Can you tell me the exact steps/noble calls you are making? 
2. Can you try using this: https://github.com/stoprocent/noble/blob/main/examples/peripheral-explorer.ts 

I have a test loop at home that is calling this script `peripheral-explorer.ts` for 4 days, and it has made 100% successful connections (~50000 so far)

stoprocent, 2025-04-19:
I think `2.1.2` could potentially fix the issue give it a go

stoprocent, 2025-04-22:
Closing for now. I beliv