stoprocent/noble #16: Disconnect on peripheral should clear all outstanding events

Apollon77, 2025-01-02:
I got an issue with duplicate code execution when a disconnect happens while being in "connect logic".
Here is what happens

* Connect 1 is send and "connected " event is emitted
* Then we send discoverServicesAsync which registers a "once" event handler to get the result ... this now does not return ... BUT a disconnect happens. means "disconnect" event is emitted. But in fact the expected command is never errored or such and so also this promise is not resolved!
* the disconnect event leads to a retry now in my code, means we get the newest peripheral (ok in this case it was not replaced btw) and connect. Connect is emitted
* We send the discoverServicesAsync again ... which leads to a second "once" event listener.
* And NOW it works and the event is emitted to both handlers and response resolves BOTH promises - the initial one and also the new one ... and so the code in "connected" runs further twice

So a disconnect should clear all outstanding events on the peripheral that might be and ideally also fail all promises

dwimberger, 2025-02-26:
We have observed the same issue with disconnects during  `discoverSomeServicesAndCharacteristicsAsync()`.
`const { characteristics } = await this.device.discoverSomeServicesAndCharacteristicsAsync(services, characteristicsIdentifiers);`

If a device disconnect happens during this procedure, it does never return/waits forever.
This can be reproduced with 1.18.2, 1.18.3 and 1.19.0. All versions are pinned to a 1.4.x hci-socket version:
"overrides": {
    "@stoprocent/bluetooth-hci-socket": "1.4.x"
 }

One thing that we have seen is, that there are no callbacks if not all services where found:
`        
if (numDiscovered === services.length) {
   if (callback) {
     callback(null, services, allCharacteristics);
    }
}
`

However in some instances the disconnect does not receive a callback because the event `servicesDiscover` seems not to be issued, but no other is handled:
`
this.once('servicesDiscover', services => {
      callback(null, services);
});
`
Still trying to analyse this one better and will follow up if I find out more.

stoprocent, 2025-02-26:
Are you calling disconnect while `discoverSomeServicesAndCharacteristicsAsync` or is it happening for some other reason?
I understand, and yes, looking at the code, it could be the case. There is even a comment done by the original author `// TODO: handle `error`?`

This is something that is easy to fix

dwimberger, 2025-02-26:
No, the disconnect happens for other reasons and while talking to a real device.

We have found some rather ugly workarounds to get the events handled. But I am still cleaning up. It would maybe be good to discuss it, before proposing it as a patch.

stoprocent, 2025-04-03:
Will fix it with today's release

dwimberger, 2025-04-04:
We have had our changes working well for a while now.

Should we create a PR? Or will you go a different route?

stoprocent, 2025-04-18:
Hey @dwimberger  and @Apollon77, mind checking the referenced commit to see if the test covers your use case properly?

closed 2025-04-22 by stoprocent