stoprocent/noble #18: Subscribe on Matter-C2 (UUID = 18EE2EF5-263D-4559-959F-4F9C429F9D12) is not working for a device (Shelly Smart Plug)

Apollon77, 2025-01-04:
I subscribe to C2 and when I interpret the data right it also receves data but the event is not triggered in this case

```
2025-01-04 17:17:49.038 DEBUG  BleChannel           Connect to Peripheral now (try 1)
  hci create le conn - writing: 010d2019600030000000569994eabf8c000600120000002a0004000600 +9ms
  hci onSocketData: 0111200700569994eabf8c +460ms
  hci 	event type = 1 +1ms
  hci 		cmd = 8209 +0ms
  hci 		data len = 7 +0ms
  hci onSocketData: 040e0401112000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8209 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010b200700600060000001 +1ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8203 +0ms
  hci 		data len = 7 +0ms
  hci onSocketData: 040e04010b2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  noble scanParametersSet +472ms
  hci onSocketData: 010c20020101 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +0ms
  hci 			enable scanning = true +0ms
  hci 			filter duplicates = true +0ms
  noble scanStart +1ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  hci onSocketData: 043e1b02010000569994eabf8c0f0201060b16f6ff0054019014010000bf +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 27 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 0000569994eabf8c0f0201060b16f6ff0054019014010000bf +0ms
  hci 			type = 0 +0ms
  hci 			address = 8c:bf:ea:94:99:56 +1ms
  hci 			address type = public +0ms
  hci 			eir = 0201060b16f6ff0054019014010000 +0ms
  hci 			rssi = -65 +0ms
  hci onSocketData: 010c20020000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +0ms
  hci 			enable scanning = false +1ms
  hci 			filter duplicates = false +0ms
  hci onSocketData: 040e04010c2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010d2019600060000000569994eabf8c001800280000002a0000000000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8205 +1ms
  hci 		data len = 25 +0ms
  hci onSocketData: 040f0400010d20 +0ms
[... 189 lines cut ...]
  hci onSocketData: 0413050140000100 +87ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 1 +0ms
  hci flush - pending: 1 queue length: 0 +0ms
  hci onSocketData: 0413050140000100 +250ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 1 +0ms
  hci flush - pending: 0 queue length: 0 +1ms
  hci onSocketData: 04050400400013 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +0ms
  hci 		handle = 64 +0ms
  hci 		reason = 19 +0ms
  hci flush - pending: 0 queue length: 0 +1ms
```

Directly after subscribing data come in the look like the response:

>   hci onSocketData: 0240200d00090004001d0500656c04f10006 +143ms

(Matter has 656c as kind of prefix and a length of 6 bytes .. this would match to the received data here. But I do not get any callback on the subscribe).

Apollon77, 2025-01-04:
In comparism here the same log part from a device where it works (a Sonoff Matter relay). Maybe you see the reason?

```
2025-01-04 19:43:07.452 DEBUG  BleChannel           Peripheral ed:b2:64:b8:ff:8e: Connect to Peripheral now (try 1)
  hci create le conn - writing: 010d20196000300000018effb864b2ed000600120000002a0004000600 +10ms
  hci onSocketData: 01112007018effb864b2ed +2s
  hci 	event type = 1 +0ms
  hci 		cmd = 8209 +0ms
  hci 		data len = 7 +1ms
  hci onSocketData: 040e0401112000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8209 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010b200700600060000001 +0ms
  hci 	event type = 1 +1ms
  hci 		cmd = 8203 +0ms
  hci 		data len = 7 +0ms
  hci onSocketData: 040e04010b2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble scanParametersSet +2s
  hci onSocketData: 010c20020101 +1ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +0ms
  hci 			enable scanning = true +0ms
  hci 			filter duplicates = true +0ms
  noble scanStart +1ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 043e2b020100018effb864b2ed1f0201060b16f6ff00280821130200010e08636b5f3130303231393332323900cc +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 43 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 00018effb864b2ed1f0201060b16f6ff00280821130200010e08636b5f3130303231393332323900cc +0ms
  hci 			type = 0 +0ms
  hci 			address = ed:b2:64:b8:ff:8e +1ms
  hci 			address type = random +0ms
  hci 			eir = 0201060b16f6ff00280821130200010e08636b5f3130303231393332323900 +0ms
  hci 			rssi = -52 +0ms
  hci onSocketData: 010c20020000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +1ms
  hci 			enable scanning = false +0ms
  hci 			filter duplicates = false +0ms
  hci onSocketData: 040e04010c2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010d20196000600000018effb864b2ed001800280000002a0000000000 +1ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8205 +0ms
  hci 		data len = 25 +0ms
  hci onSocketData: 040f0400010d20 +0ms
[... 143 lines cut ...]
  hci write acl data packet - writing: 024000090005000400121a000200 +0ms
  hci onSocketData: 0413050140000200 +94ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 2 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 02402005000100040013 +48ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +1ms
  hci 		data = 13 +0ms
  att ed:b2:64:b8:ff:8e: read: 13 +145ms
  hci onSocketData: 0240200d00090004001d1900656c04f10005 +2ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 1d1900656c04f10005 +1ms
  att ed:b2:64:b8:ff:8e: write: 1e +4ms
2025-01-04 19:43:09.986 INFO   BleChannel           Peripheral ed:b2:64:b8:ff:8e: Received Matter handshake response: 656c04f10005.
  hci push to acl queue: 0240000500010004001e +4ms
  hci flush - pending: 0 queue length: 1 +0ms
2025-01-04 19:43:09.989 DEBUG  BtpSessionHandler    Handshake request version: 4 attMtu: 241 windowSize: 5

```

stoprocent, 2025-04-18:
I did a lot of updates and fixes for GATT in #24 you will have to re-test.

Apollon77, 2025-04-18:
Alright. Would it be possible to get an npm release (maybe under a @dev tag on npm or such?) so that I can directly test it in the application with an "official" build?

Apollon77, 2025-04-19:
Ok with 2.0.1 I have this with an other device but looks same-ish

```
2025-04-19 11:40:54.642 DEBUG  NobleBleClient       Found peripheral f1:ef:34:2a:2f:e6 (undefined): {"localName":"undefined","txPowerLevel":"undefined","manufacturerData":"undefined","serviceData":[{"uuid":"fff6","data":{"type":"Buffer","data":[0,135,9,241,255,4,128,1]}}],"serviceUuids":[],"serviceSolicitationUuids":[]}
2025-04-19 11:40:54.643 DEBUG  BleScanner           Discovered device f1:ef:34:2a:2f:e6 008709f1ff048001
2025-04-19 11:40:54.644 DEBUG  BleScanner           Re-Discovered device f1:ef:34:2a:2f:e6 data: {"deviceIdentifier":"f1:ef:34:2a:2f:e6","D":2439,"SD":9,"VP":"65521+32772","CM":1,"addresses":[{"type":"ble","peripheralAddress":"f1:ef:34:2a:2f:e6"}]}
  hci onSocketData: 010c20020000 +5ms
  hci 	event type = 1 +1ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +0ms
  hci 			enable scanning = false +0ms
  hci 			filter duplicates = false +0ms
  hci onSocketData: 040e04010c2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  hci onSocketData: 010d2019600060000001e62f2a34eff1001800280000002a0000000000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8205 +0ms
  hci 		data len = 25 +0ms
  hci onSocketData: 040f0400010d20 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +0ms
  hci 		cmd = 8205 +1ms
  hci onSocketData: 043e13010040000001e62f2a34eff1270000002a0000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 1 +0ms
  hci 		LE meta event data length = 19 +0ms
  hci 		LE meta event num reports = 0 +0ms
  hci 		LE meta event data = 40000001e62f2a34eff1270000002a0000 +0ms
  hci 			handle = 64 +1ms
  hci 			role = 0 +0ms
  hci 			address type = random +0ms
  hci 			address = f1:ef:34:2a:2f:e6 +0ms
  hci 			interval = 48.75 +1ms
  hci 			latency = 0 +0ms
  hci 			supervision timeout = 420 +0ms
  hci 			master clock accuracy = 0 +0ms
  att f1:ef:34:2a:2f:e6: write: 020001 +0ms
  hci push to acl queue: 024000070003000400020001 +7ms
  hci flush - pending: 0 queue length: 1 +0ms
  hci write acl data packet - writing: 024000070003000400020001 +1ms
  hci onSocketData: 011620024000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8214 +0ms
  hci 		data len = 2 +0ms
  hci onSocketData: 040f0400011620 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +1ms
  hci 		cmd = 8214 +0ms
  hci onSocketData: 043e0c040040002d00000000000000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 4 +0ms
  hci 		LE meta event data length = 12 +0ms
  hci 		LE meta event num reports = 0 +0ms
  hci 		LE meta event data = 40002d00000000000000 +0ms
  hci onSocketData: 02402007000300040003f700 +185ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +1ms
  hci 		data = 03f700 +0ms
  att f1:ef:34:2a:2f:e6: read: 03f700 +193ms
  att f1:ef:34:2a:2f:e6: new MTU is 247 +0ms
[... 222 lines cut ...]
2025-04-19 11:41:00.318 DEBUG  BleChannel           Peripheral f1:ef:34:2a:2f:e6: Connect to Peripheral now (try 2)
  hci set scan enabled - writing: 010c20020001 +29ms
2025-04-19 11:41:00.320 ERROR  BleChannel           Peripheral f1:ef:34:2a:2f:e6: Error while unsubscribing 22
  hci onSocketData: 040e04010c200c +11ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
2025-04-19 11:41:04.490 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 12 seconds
2025-04-19 11:41:09.944 DEBUG  MdnsServer           Expiring records instanceName: D5289F5F8EA64462-76DE0DCD69288F24._matter._tcp.local port: 40537 netInterface: wlan0
2025-04-19 11:41:09.954 DEBUG  MdnsScanner          Removing operational device D5289F5F8EA64462-76DE0DCD69288F24._matter._tcp.local from cache (interface wlan0) because of ttl=0
  hci set scan enabled - writing: 010c20020001 +10s

```

Also here I see the data incoming but my code does not get a callback call:

```
  hci onSocketData: 0240200d00090004001d1000656c04f10005 +1ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +1ms
  hci 		data = 1d1000656c04f10005 +0ms
```

Apollon77, 2025-04-19:
Also a connect after a disconnect in such case does not work (no idea if related?):

```
2025-04-19 12:42:46.465 DEBUG  BleChannel           Peripheral d6:21:47:47:5f:57: Handshake Response not received. Disconnect from peripheral
  hci disconnect - writing: 01060403400013 +3ms
  hci onSocketData: 040f0400010604 +2ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +1ms
  hci 		status = 0 +0ms
  hci 		cmd = 1030 +0ms
  hci onSocketData: 04050400400013 +84ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +0ms
  hci 		handle = 64 +1ms
  hci 		reason = 19 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
2025-04-19 12:42:46.559 ERROR  BleChannel           Peripheral d6:21:47:47:5f:57 disconnected while trying to connect, try again Peripheral d6:21:47:47:5f:57: Handshake Response not received
  at StandardTimer.callback (/home/ingof/official-matter.js/packages/nodejs-ble/src/NobleBleChannel.ts:429:31)
2025-04-19 12:42:46.580 DEBUG  BleChannel           Peripheral d6:21:47:47:5f:57: Connect to Peripheral now (try 2)
  hci set scan enabled - writing: 010c20020001 +26ms
  hci onSocketData: 040e04010c200c +12ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
```

basically ble is "blocked" in the nodejs process then ... and all tries to "connectAsync" hang forever and promises never return.

stoprocent, 2025-04-19:
Good catch! Your device is sending an Indication, not a simple Notification, which requires explicit acknowledgment. I'll have a closer look at handling this properly. To be honest, I've been primarily using plain Notifications to avoid timing complications, but I'll mock something up to test this scenario and verify. 

One thing I see is that after that incoming message, there is a confirmation sent:

```
 att f1:ef:34:2a:2f:e6: write: 1e +2ms
```

which is line `113` in `gatt.js`

```
if (data[0] === ATT_OP_HANDLE_IND) {
    this._queueCommand(this.handleConfirmation(), null, () => {
        this.emit('handleConfirmation', this._address, valueHandle);
    });
}
```

Can you put a console log somewhere inside the callback before so we can see it event is being triggered

```
this.emit('handleConfirmation', this._address, valueHandle);
```

[image]

Apollon77, 2025-04-19:
Alright I will do later. In fact the above device (just tested) work btw with 1.x noble version!

stoprocent, 2025-04-19:
Sorry, I forgot to mention an important change, and I'll push an update shortly. Previously, there was confusion around characteristic events because the old noble emitted two separate events: `read` and `data`. I've simplified this since it didn't make much sense. In most cases (99%), users were just listening to the `data` event. That event is now removed, so please switch to listening on the `read` event and try again.

Apollon77, 2025-04-19:
Alright I adjusted that and will now also add the logline

Apollon77, 2025-04-19:
Ok ... after changing data -> read in listening I have 
* error 62 on connect often ... I will update the respective issue
... but I also had success cases when connect was working

stoprocent, 2025-04-19:
Sorry for the confusion! I'm planning to push an update that uses the `data` event instead of `read`. The main reason is that `read` was marked as deprecated in the original library, so this change aligns better with the intended usage. I’ll make sure to clearly document this change in the README to avoid any confusion going forward.

Apollon77, 2025-04-19:
Ahhh ... ok but thats ok ... as said in 2.0.1 with using "read" it all works IF not connect error 62 happens (which unfortunately is 80% of the cases)

stoprocent, 2025-04-19:
Ok back to `data` on 2.1.0 :) Sorry for confusion.
I have also added very clear guidelines now: https://github.com/stoprocent/noble?tab=readme-ov-file#characteristic-methods

stoprocent, 2025-04-19:
Can we close it ?

Apollon77, 2025-04-19:
nope... I tested the original shelly device (for what I originally have opened the issue) too with noble 2.0.1 and this one still does not see the handshake response

```
2025-04-19 17:37:36.299 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Handle characteristic: 18ee2ef5263d4559959f4f9c429f9d11 write
2025-04-19 17:37:36.301 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Found C1 characteristic
2025-04-19 17:37:36.302 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Handle characteristic: 18ee2ef5263d4559959f4f9c429f9d12 indicate
2025-04-19 17:37:36.303 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Found C2 characteristic
2025-04-19 17:37:36.305 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Using MTU=244 bytes (Peripheral supports up to 500 bytes)
2025-04-19 17:37:36.307 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Sending BTP handshake request: "656c04000000f400ff"
  att 8c:bf:ea:94:99:56: write: 120300656c04000000f400ff +13ms
  hci push to acl queue: 02400010000c000400120300656c04000000f400ff +14ms
  hci flush - pending: 1 queue length: 1 +0ms
  hci write acl data packet - writing: 02400010000c000400120300656c04000000f400ff +1ms
  hci onSocketData: 0413050140000200 +44ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 2 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 02402005000100040013 +38ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 13 +0ms
  att 8c:bf:ea:94:99:56: read: 13 +84ms
2025-04-19 17:37:36.395 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Subscribing to C2 characteristic
  att 8c:bf:ea:94:99:56: write: 08040006000229 +5ms
  hci push to acl queue: 0240000b000700040008040006000229 +5ms
  hci flush - pending: 0 queue length: 1 +0ms
  hci write acl data packet - writing: 0240000b000700040008040006000229 +1ms
  hci onSocketData: 0240200a0006000400090406000000 +91ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 090406000000 +1ms
  att 8c:bf:ea:94:99:56: read: 090406000000 +93ms
  att 8c:bf:ea:94:99:56: write: 1206000200 +1ms
  hci push to acl queue: 0240000900050004001206000200 +1ms
  hci flush - pending: 1 queue length: 1 +0ms
  hci write acl data packet - writing: 0240000900050004001206000200 +1ms
  hci onSocketData: 0413050140000200 +46ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 2 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 0240200d00090004001d0500656c04f10006 +48ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 1d0500656c04f10006 +0ms
  hci onSocketData: 02402005000100040013 +2ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 13 +0ms
  att 8c:bf:ea:94:99:56: read: 13 +99ms
  att 8c:bf:ea:94:99:56: write: 1e +2ms
  hci push to acl queue: 0240000500010004001e +3ms
  hci flush - pending: 0 queue length: 1 +0ms
  hci write acl data packet - writing: 0240000500010004001e +0ms
  hci onSocketData: 0413050140000100 +275ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 1 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
2025-04-19 17:37:38.860 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 6 seconds
  att 8c:bf:ea:94:99:56: write: 08040006000229 +5s
  hci push to acl queue: 0240000b000700040008040006000229 +4s
  hci flush - pending: 0 queue length: 1 +0ms
  hci write acl data packet - writing: 0240000b000700040008040006000229 +0ms
  hci onSocketData: 0240200a0006000400090406000200 +70ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 090406000200 +0ms
  att 8c:bf:ea:94:99:56: read: 090406000200 +73ms
  att 8c:bf:ea:94:99:56: write: 1206000000 +0ms
  hci push to acl queue: 0240000900050004001206000000 +1ms
  hci flush - pending: 1 queue length: 1 +0ms
  hci write acl data packet - writing: 0240000900050004001206000000 +0ms
  hci onSocketData: 0413050140000200 +34ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 2 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 04050400400013 +244ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +1ms
  hci 		handle = 64 +0ms
  hci 		reason = 19 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
2025-04-19 17:37:41.663 ERROR  BleChannel           Peripheral 8c:bf:ea:94:99:56: Error while unsubscribing 19
2025-04-19 17:37:41.665 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Handshake Response not received. Disconnect from peripheral
  hci disconnect - writing: 01060403000013 +9ms
  hci onSocketData: 040f0402010604 +34ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 2 +0ms
  hci 		cmd = 1030 +1ms

```

stoprocent, 2025-04-19:
ok looking at your log:
```
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 04050400400013 +244ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +1ms
  hci 		handle = 64 +0ms
  hci 		reason = 19 +0ms
```

Sub-event is `disconnect` and reason in `0x13` or `19` in dec - `Remote User Terminated Connection`, meaning it's a device that has terminated the connection.

Apollon77, 2025-04-19:
Yes it has disconnected because we did not reacted on the send message before

basically 

```
 hci onSocketData: 0240200d00090004001d0500656c04f10006 +48ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +0ms
  hci 		data = 1d0500656c04f10006 +0ms
```

Was not reaching us and so the device disconnected

stoprocent, 2025-04-19:
this is missing notification. I dont think there is an issue here tbh.

Apollon77, 2025-04-19:
Ok here updated Shelly log:

```
  hci 			simul = 0 +0ms
  hci onSocketData: 040e0701022000fb0007 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8194 +0ms
  hci 		status = 0 +0ms
  hci 		result = fb0007 +0ms
  hci le buffer size: length = 251, num = 7 +0ms
  hci onSocketData: 040e0a01091000efe53e32a6dc +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +0ms
  hci 		result = efe53e32a6dc +0ms
  hci address = dc:a6:32:3e:e5:ef +0ms
  noble addressChange dc:a6:32:3e:e5:ef +5ms
  hci onSocketData: 040e04010c200c +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e04010b2000 +0ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble stateChange poweredOn +3ms
2025-04-19 23:35:59.215 DEBUG  NobleBleClient       Noble state changed to poweredOn
  noble scanParametersSet +2ms
  hci onSocketData: 040e04010c200c +3ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e04010b2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble scanParametersSet +2ms
2025-04-19 23:35:59.235 INFO   CertificateAuthority Loaded stored credentials with ID 0
2025-04-19 23:35:59.243 INFO   MatterController     Used existing fabric
2025-04-19 23:35:59.260 INFO   SessionManager       restoring resumption record for node 1500 and peer node 1500 for fabric index 1
2025-04-19 23:35:59.261 INFO   SessionManager       restoring resumption record for node 1100 and peer node 1100 for fabric index 1
2025-04-19 23:35:59.263 INFO   SessionManager       restoring resumption record for node 1300 and peer node 1300 for fabric index 1
2025-04-19 23:35:59.264 INFO   SessionManager       restoring resumption record for node 1200 and peer node 1200 for fabric index 1
2025-04-19 23:35:59.265 INFO   SessionManager       restoring resumption record for node 3000 and peer node 3000 for fabric index 1
2025-04-19 23:35:59.265 INFO   SessionManager       restoring resumption record for node 2400 and peer node 2400 for fabric index 1
2025-04-19 23:35:59.266 INFO   SessionManager       restoring resumption record for node 4100 and peer node 4100 for fabric index 1
2025-04-19 23:35:59.267 INFO   SessionManager       restoring resumption record for node 2000 and peer node 2000 for fabric index 1
2025-04-19 23:35:59.267 INFO   SessionManager       restoring resumption record for node 1000 and peer node 1000 for fabric index 1
2025-04-19 23:35:59.269 INFO   SessionManager       restoring resumption record for node 1400 and peer node 1400 for fabric index 1
2025-04-19 23:35:59.269 INFO   SessionManager       restoring resumption record for node 2600 and peer node 2600 for fabric index 1
2025-04-19 23:35:59.271 INFO   SessionManager       restoring resumption record for node 5000 and peer node 5000 for fabric index 1
2025-04-19 23:35:59.672 DEBUG  DeviceAdvertiser     Announcing fabricIndex: 1 fabricId: 1
2025-04-19 23:35:59.680 DEBUG  MdnsBroadcaster      Announcement Generator: Fabric id: D5289F5F8EA64462-76DE0DCD69288F24 qname: D5289F5F8EA64462-76DE0DCD69288F24._matter._tcp.local port: 51797 interface: wlan0
2025-04-19 23:35:59.691 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: 0.0.0.0:5353
2025-04-19 23:35:59.693 DEBUG  NodejsChannel        Socket created and bound remoteAddress: undefined:5353 localAddress: :::5353
2025-04-19 23:35:59.695 DEBUG  NodejsChannel        Initialize multicast address: 192.168.178.41:5353 interface: wlan0 type: udp4
2025-04-19 23:35:59.697 DEBUG  NodejsChannel        Initialize multicast address: ::%wlan0:5353 interface: wlan0 type: udp6
2025-04-19 23:35:59.845 DEBUG  MdnsScanner          Adding operational device D5289F5F8EA64462-76DE0DCD69288F24._matter._tcp.local in cache (interface wlan0, ttl=120) with TXT data: SII: 500 SAI: 300 SAT: 4000 T: 0 ICD: 0
2025-04-19 23:35:59.848 DEBUG  MdnsScanner          Added 4 IPs for operational device D5289F5F8EA64462-76DE0DCD69288F24._matter._tcp.local to cache (interface wlan0): type: udp ip: 2a02:8071:2c87:399d:dea6:32ff:fe3e:e5ee port: 51797 type: udp ip: fdb8:15eb:fc0f::dea6:32ff:fe3e:e5ee port: 51797 type: udp ip: fe80::dea6:32ff:fe3e:e5ee%wlan0 port: 51797 type: udp ip: 192.168.178.41 port: 51797
{"discovery":{"knownAddress":"undefined","identifierData":{"shortDiscriminator":1},"discoveryCapabilities":{"ble":true,"onIpNetwork":true}},"passcode":53616041,"commissioning":{"nodeId":"1700","regulatoryLocation":1,"regulatoryCountryCode":"XX"}}
[... 498 lines cut ...]
  hci flush - pending: 1 queue length: 1 +1ms
  hci write acl data packet - writing: 0240000900050004001206000000 +0ms
  hci onSocketData: 0413050140000200 +46ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 19 +0ms
  hci 	handle = 64 +0ms
  hci 		completed = 2 +1ms
  hci flush - pending: 0 queue length: 0 +0ms
  hci onSocketData: 04050400400013 +243ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +1ms
  hci 		handle = 64 +0ms
  hci 		reason = 19 +0ms
  hci flush - pending: 0 queue length: 0 +0ms
[... cut ...]
2025-04-19 23:36:07.318 DEBUG  BleChannel           Peripheral 8c:bf:ea:94:99:56: Handshake Response not received. Disconnect from peripheral
  hci disconnect - writing: 01060403000013 +7ms
  hci onSocketData: 040f0402010604 +30ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 2 +0ms
  hci 		cmd = 1030 +1ms
```

The same code successfully works with 3 other matter devcies, so the general notification / "read" event handling and such is working. But here for shelly this is not triggered somehow

Apollon77, 2025-04-24:
[shelly-noworking.txt](https://github.com/user-attachments/files/19890435/shelly-noworking.txt)
[working-espressif.txt](https://github.com/user-attachments/files/19890436/working-espressif.txt)

stoprocent, 2025-04-24:
Actually, it wasn't a problem, and we found the issue elsewhere :) @Apollon77 I'm closing it. If `noble.stop` will be an issue that's unrelated to this one.

closed 2025-04-24 by stoprocent

Apollon77, 2025-04-25:
Exactly, but also here for referwence: The above described behavior of shellies mean that it seems that the first value of a subscriptioin was sent already before the subscription itself was acked.

So that basically means: Register for the "data" event BEFORE you call "await subscribeAsync()" because if you do afterwards it could be already too late