stoprocent/noble #19: (Re-) discovered devices should reset peripheral status?

Apollon77, 2025-01-04:
When I read corrrectly a peripheral which is connected can not send "connectable announcements". If this is true then a re-discovery of a device which is connectable should reset the peripheral state to "disconnected". Or!?
This could clean strange intermediate states like from #17

stoprocent, 2025-01-05:
Please provide steps to reproduce this issues you have created are very lazy. Will re-open when proper steps to reproduce are provided

Apollon77, 2025-01-05:
Thats not that easy ...
In fact I tested a lot and had the effect that after doing connect/disconnect on a device and trying to do connect again this new connect call was not working and "hanging" (promise never got resolved). In this case the peripheral had the status "connecting". Even after cancelConnect() (see other issue) this state stayed "connecting".

I easiely could start a new scanning for device and such and when the device was then rediscovered the status stayed "connecting". But I was - after the rediscovery - able to connect to the device. After this the state was updated.

I mainly refer to 

https://github.com/stoprocent/noble/blob/main/lib/noble.js#L227-L245

When the peripheral which is discovered is already known some values get updated, but the peripheral state stays untouched. So my main thing was if my assumptions above are correct (about the only connectable or discoverable) then also setting peripheral.status="disconnected" should be ok here in the case of existing peripherals

Apollon77, 2025-01-05:
Ahh check the log in the main post at #13 ... there at the very end is such a "connect" that never resolved ... after this the peripheral has this "connecting" state and will also stay that ways when a new scan re-discovers the device because the state is not "re-set" on discovery and the peripheral is already known

stoprocent, 2025-01-05:
Can you clean up issues because im lost a bit. You dropped suddenly 7 issues and somehow they are all related. Please clean it up

stoprocent, 2025-04-03:
> When I read corrrectly a peripheral which is connected can not send "connectable announcements". If this is true then a re-discovery of a device which is connectable should reset the peripheral [...]

No this is not true. Peripheral can accept multiple connections. Newest `bleno` supports this. But there could be something with statuses I will check this.

stoprocent, 2025-04-18:
@Apollon77 I think this will not happen any more but next time when you create an issue please do so with easy steps to reproduce. Have a look at #23 this is excellent example.