stoprocent/noble #20: Matter device not discovered on windows

spudwebb, 2025-01-08:
Using the matter.js library, some matter devices are not discovered on windows. I can see all of them when using the examples\advertisement-discovery.js script from noble but some of them are not passed back to the matter.js lib.

here is for example one that is correctly passed to matter.js:

```
Peripheral discovered (30de4bdb940d with address <30:de:4b:db:94:0d, public>, connectable: true, scannable: undefined, RSSI -68:
        hello my local name is:
                P125M
        can I interest you in any of the following advertised services:
                ["8641","fff6"]
        here is my service data:
                "fff6": "00bc0d8811010101"
        here is my manufacturer data:
                ""
        my TX power level is:
                0
```

and one that isn't:
```
Peripheral discovered (d2961048b505 with address <d2:96:10:48:b5:05, random>, connectable: true, scannable: undefined, RSSI -62:
        hello my local name is:

        can I interest you in any of the following advertised services:
                []
        here is my service data:
                "fff6": "00e30b0a13530001"
        here is my manufacturer data:
                ""
        my TX power level is:
                0
```

Is it because the advertised services or local name is missing?
For comparison the same device is correctly discovered by matter.js on Linux and here is what I get on Linux with the advertisement-discovery.js script:

```
Peripheral discovered (d2961048b505 with address <d2:96:10:48:b5:05, random>, connectable: true, scannable: false, RSSI -84:
        hello my local name is:
                EveEnergy51BE3
        can I interest you in any of the following advertised services:
                []
        here is my service data:
                "fff6": "00e30b0a13530001"
```

Apollon77, 2025-01-09:
Interesting ... it has service specific additional data but the service is not in the list of announced services

stoprocent, 2025-01-09:
@spudwebb I guess if you see it in `examples\advertisement-discovery.js` then it's not a noble issue but `matter.js`

stoprocent, 2025-01-09:
> Interesting ... it has service specific additional data but the service is not in the list of announced services

that is fine with BLE spec list of advertised service's is one thing and service data is another thing.

Apollon77, 2025-01-09:
But then it is an issue with the device not announcing a matter service? Sure I can add a check if service data are there but service is not announced ... still feels strange

Apollon77, 2025-01-09:
Ahh ok ... currently we only discover services that also announce matter service so we will never discover these devices

>             await noble.startScanningAsync([BLE_MATTER_SERVICE_UUID], true);

this is what we do ... which means the devices which do not announce the service even with having service specific data are never returned by noble.

Apollon77, 2025-01-09:
Hm ... Ok I look through noble code a bit ... 

* in hci bindings (linux) the diswcovery is discovery without filters and the onDiscover() combines service data ids and announced ids  when it comes to the question if it should be emitted
* in windows it seems to me that it is just filtered by the service id?

Could that be? Then it would describe the different behavior

stoprocent, 2025-04-03:
You're right — the behavior was inconsistent between platforms. On Linux, noble combines service UUIDs from both the advertised service list (complete/incomplete/solicited) and the service data when deciding whether to emit a discover event. On Windows, it was relying solely on the advertised service list, which explains why the device wasn’t discovered there when no services were advertised.

While I agree this might not be the cleanest approach overall, today’s release includes a patch that makes the behavior consistent across platforms. Now, when you specify a service UUID, discovery will be filtered using both the advertised UUID lists and the service data UUIDs — on all platforms.

closed 2025-04-03 by stoprocent