stoprocent/noble #21: No disconnectAsync() function on Nible

Abasz, 2025-01-19:
According to the exported typescript typing file there should be a disconnectAsync() function. But in the noble.js file there isnt any.

So disconnect is only possible with the normal .disconnect() function

Abasz, 2025-01-19:
Actually, the normal disconnect function signature is also wrong as the typeing file specifies a callback, but the code expects UUID.

stoprocent, 2025-01-19:
Sorry, Im super loaded with work can you create a pull-request? I will have a look into this. Seems like super easy thing

Abasz, 2025-01-20:
Yes on the peripheral its fine but not directly on the noble import:

[image]

This is the call signature of noble disconnect: https://github.com/stoprocent/noble/blob/ecb43339054cd3ebc2f5ff3da2600eb67c8e3fe9/lib/noble.js#L296

and the peripheral expects something different. 

I will have a look at some point, its not really a breaking issue.

stoprocent, 2025-01-20:
I'm a bit lost tbh. You should use disconnect from peripheral and not noble directly. What is the use case?
Noble disconnect is the same thing as peripheral disconnect. You have a peripheral at your screenshot, so why don't you call it there on peripheral?

Abasz, 2025-01-20:
Technically I call both :) But I thought noble.disconnect (without any handler) would disconnect all peripherals. This is why I originally wanted to call noble.disconnect() only (because then I dont need to track the peripherals locally). 

```js
    await _peripheral.disconnectAsync()
    noble.disconnect()
```

But if you are saying that noble.disconnect() does not work as I expect, then this is irrelevant indeed.

What I want is to clean up noble but without tearing down the Hci socket (or reset). This relates to OpenRowingMonitor issue where noble and bleno are used simultaneously.

stoprocent, 2025-01-21:
No `noble.disconnect(...)` requires peripheral UDID. For that issue with `OpenRowingMonitor` I think my work around making bleno and noble work together will solve the problem. I've already made a couple of changes to make this happen.

closed 2025-01-21 by stoprocent