stoprocent/noble #22: HCI Socket connections sometimes fail

dwimberger, 2025-03-01:
**Problem description**

The symptom is always related to a disconnect after first establishing a connection successfully. 
The Node process gets stuck or dies with the following error:

`
  Peripheral Failed to discover services +1ms
  Peripheral Device could not connect at 2025-03-01T09:32:29.765+01:00 +0ms
  Bridge connecting to device with retries: 4 +2m
/home/users/me/ble-bridge/node_modules/@stoprocent/noble/lib/hci-socket/hci.js:745
    this._aclConnections.get(handle).pending++;
                         ^

TypeError: Cannot read properties of undefined (reading 'pending')
    at Hci.flushAcl (/Users/dieter/Documents/Development/javascript/nwp-pi-bridge/node_modules/@stoprocent/noble/lib/hci-socket/hci.js:745:26)
`

**Environment**
Node 20 LTS and Node 22 LTS (happens with both)

`
"@stoprocent/bluetooth-hci-socket": "1.5.2" (override)
"@stoprocent/noble": "1.19.1",
`

We are using [this patch](https://github.com/stoprocent/noble/issues/16#issuecomment-2685581659).
We are using this setting for a known serialport bug that we have run into:
`
// https://github.com/serialport/node-serialport/issues/2735
process.env.UV_USE_IO_URING = 0
`

[node-bluetooth-hci-socket](https://github.com/stoprocent/node-bluetooth-hci-socket) is configured to work with a Nrf52840 Dongle with the Zephyr HCI Sample configured as described in the documentation (updated to latest zephyr version though).

We are intentionally using some "hard to reach" peripherals to investigate the stability under "real world" conditions.

stoprocent, 2025-03-01:
Does the patch solves the issue ?

stoprocent, 2025-03-01:
I need to review all these `this.once`. Im not huge fan of them, especially that in many cases under heavy load of message this is causing max listeners error.

dwimberger, 2025-03-02:
No, the patch does not solve this issue in particular. It seems to be a connection loss that is happening right after establishing the connection, before further discovery of services and characteristics.

It does however, solve the issue that we return from cases where we have unsuccessful ```discoverService``` and ```discoverCharacteristics``` due to connection loss. 

I will look into it at the HCI layer, because it may also still be something related to the firmware.
Also maybe it makes sense to bring part of the noble implementation to an async await world for better readability.

stoprocent, 2025-04-18:
Will you be able to retest after #24 is merged? There are so many changes and I've added some new APIs that it's hard to relate now.

stoprocent, 2025-04-27:
@dwimberger bump 👍

stoprocent, 2025-05-20:
Closing due to inactivity

closed 2025-05-20 by stoprocent