stoprocent/noble #26: Can not connect to a Espressif TBR board with chip example

Apollon77, 2025-04-19:
I used a Thread Border Router Espressif board and flashed the chip OTBR test app on it. Trying to connect from a RPI to it does not work

```
2025-04-19 12:51:34.719 DEBUG  NobleBleClient       Found peripheral d5:e1:a6:c3:6f:9e (undefined): {"localName":"undefined","txPowerLevel":"undefined","manufacturerData":"undefined","serviceData":[{"uuid":"fff6","data":{"type":"Buffer","data":[0,0,15,241,255,0,128,0]}}],"serviceUuids":[],"serviceSolicitationUuids":[]}
2025-04-19 12:51:34.721 DEBUG  BleScanner           Discovered device d5:e1:a6:c3:6f:9e 00000ff1ff008000
2025-04-19 12:51:34.722 DEBUG  BleScanner           Discovered device d5:e1:a6:c3:6f:9e data: {"deviceIdentifier":"d5:e1:a6:c3:6f:9e","D":3840,"SD":15,"VP":"65521+32768","CM":1,"addresses":[{"type":"ble","peripheralAddress":"d5:e1:a6:c3:6f:9e"}]}
2025-04-19 12:51:34.724 DEBUG  BleScanner           Finishing waiter for query D:3840, resolving: true
2025-04-19 12:51:34.725 DEBUG  NobleBleClient       Stop BLE scanning for Matter Services ...
  hci set scan enabled - writing: 010c20020001 +10ms
  hci onSocketData: 043e2a0201000064e0a972d2541e0201061aff4c000215a92ee200550111e4916c0800200c9a663ba9e064c8ce +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +1ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event data length = 42 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 000064e0a972d2541e0201061aff4c000215a92ee200550111e4916c0800200c9a663ba9e064c8ce +0ms
  hci 			type = 0 +0ms
  hci 			address = 54:d2:72:a9:e0:64 +0ms
  hci 			address type = public +0ms
  hci 			eir = 0201061aff4c000215a92ee200550111e4916c0800200c9a663ba9e064c8 +0ms
  hci 			rssi = -50 +0ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble scanStop +114ms
2025-04-19 12:51:34.730 INFO   ControllerDiscovery  Found 1 devices using identifier {"longDiscriminator":3840}
2025-04-19 12:51:34.732 DEBUG  ControllerDiscovery  Server addresses to try: ble://d5:e1:a6:c3:6f:9e
2025-04-19 12:51:34.733 DEBUG  ControllerDiscovery  Try to communicate with ble://d5:e1:a6:c3:6f:9e ...
2025-04-19 12:51:34.734 INFO   PeerCommissioner     Establish PASE to device VP: 65521+32768
2025-04-19 12:51:34.737 DEBUG  BleChannel           Peripheral d5:e1:a6:c3:6f:9e: Connect to Peripheral now (try 1)
  hci create le conn - writing: 010d20196000300000019e6fc3a6e1d5000600120000002a0004000600 +10ms
  hci onSocketData: 01112007019e6fc3a6e1d5 +1s
  hci 	event type = 1 +1ms
  hci 		cmd = 8209 +0ms
  hci 		data len = 7 +0ms
  hci onSocketData: 040e0401112000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8209 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010b200700600060000001 +1ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8203 +0ms
  hci 		data len = 7 +0ms
  hci onSocketData: 040e04010b2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  noble scanParametersSet +1s
  hci onSocketData: 010c20020101 +0ms
  hci 	event type = 1 +1ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +0ms
  hci 			enable scanning = true +0ms
  hci 			filter duplicates = true +1ms
  noble scanStart +2ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  hci onSocketData: 043e2b020100019e6fc3a6e1d51f0201060b16f6ff00000ff1ff00800000000000000000000000000000000000c8 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +1ms
  hci 		LE meta event data length = 43 +0ms
  hci 		LE meta event num reports = 1 +0ms
  hci 		LE meta event data = 00019e6fc3a6e1d51f0201060b16f6ff00000ff1ff00800000000000000000000000000000000000c8 +0ms
  hci 			type = 0 +0ms
  hci 			address = d5:e1:a6:c3:6f:9e +0ms
  hci 			address type = random +1ms
  hci 			eir = 0201060b16f6ff00000ff1ff00800000000000000000000000000000000000 +0ms
  hci 			rssi = -56 +0ms
  gap invalid EIR data, length = 0 +1s
  hci onSocketData: 010c20020000 +0ms
  hci 	event type = 1 +1ms
  hci 		cmd = 8204 +0ms
  hci 		data len = 2 +0ms
  hci 			LE enable scan command +1ms
  hci 			enable scanning = false +0ms
  hci 			filter duplicates = false +0ms
  hci onSocketData: 040e04010c2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 010d20196000600000019e6fc3a6e1d5001800280000002a0000000000 +1ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8205 +0ms
  hci 		data len = 25 +0ms
  hci onSocketData: 040f0400010d20 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +0ms
  hci 		cmd = 8205 +0ms
2025-04-19 12:51:36.120 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 3 seconds
2025-04-19 12:51:39.119 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 6 seconds
2025-04-19 12:51:45.119 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 12 seconds
2025-04-19 12:51:57.120 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 24 seconds
2025-04-19 12:52:04.608 DEBUG  MdnsScanner          Finishing waiter for query _L3840._sub._matterc._udp.local, resolving: true
2025-04-19 12:52:04.611 DEBUG  MdnsScanner          Removing last query _L3840._sub._matterc._udp.local and stopping announce timer
2025-04-19 12:52:04.612 INFO   ControllerDiscovery  Found 0 devices using identifier {"longDiscriminator":3840}
  hci onSocketData: 043e130100400000019e6fc3a6e1d5270000002a0000 +41s
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 1 +0ms
  hci 		LE meta event data length = 19 +0ms
  hci 		LE meta event num reports = 0 +1ms
  hci 		LE meta event data = 400000019e6fc3a6e1d5270000002a0000 +0ms
  hci 			handle = 64 +2ms
  hci 			role = 0 +0ms
  hci 			address type = random +0ms
  hci 			address = d5:e1:a6:c3:6f:9e +0ms
  hci 			interval = 48.75 +1ms
  hci 			latency = 0 +0ms
  hci 			supervision timeout = 420 +0ms
  hci 			master clock accuracy = 0 +0ms
  att d5:e1:a6:c3:6f:9e: write: 020001 +0ms
  hci push to acl queue: 024000070003000400020001 +11ms
  hci flush - pending: 0 queue length: 1 +0ms
  hci write acl data packet - writing: 024000070003000400020001 +1ms
  hci onSocketData: 011620024000 +0ms
  hci 	event type = 1 +0ms
  hci 		cmd = 8214 +0ms
  hci 		data len = 2 +0ms
  hci onSocketData: 040f0400011620 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +0ms
  hci 		cmd = 8214 +0ms
  hci onSocketData: 043e0c043e40003f00000800000000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 4 +0ms
  hci 		LE meta event data length = 12 +0ms
  hci 		LE meta event num reports = 62 +0ms
  hci 		LE meta event data = 40003f00000800000000 +0ms
  hci onSocketData: 0405040040003e +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 5 +0ms
  hci 		handle = 64 +0ms
  hci 		reason = 62 +0ms
  hci flush - pending: 0 queue length: 0 +0ms

```

stoprocent, 2025-04-19:
Youa re starting scan after sending create connection

Apollon77, 2025-04-19:
I am not doing that code wise. I saw that several times that on calling connect sometimes such a start was started already in old version

Apollon77, 2025-04-19:
Verified coe again. All places are guarded with a log and no log appears

stoprocent, 2025-04-19:
@Apollon77 closing this one and just following #13

closed 2025-04-19 by stoprocent