stoprocent/noble #27: Noble 2.0 on macos: The specified UUID is not allowed for this operation. on write

Apollon77, 2025-04-19:
No idea how to get more logs on macos, sorrym so it is just matter.js

```
2025-04-19 13:55:13.152 DEBUG  NobleBleClient       Found peripheral  (MATTER-3840): {"localName":"MATTER-3840","txPowerLevel":null,"manufacturerData":{"type":"Buffer","data":[]},"serviceData":[{"uuid":"fff6","data":{"type":"Buffer","data":[0,0,15,241,255,0,128,0]}}],"serviceUuids":[]}
2025-04-19 13:55:13.152 DEBUG  BleScanner           Discovered device  00000ff1ff008000
2025-04-19 13:55:13.153 DEBUG  BleScanner           Discovered device  data: {"deviceIdentifier":"","D":3840,"SD":15,"VP":"65521+32768","CM":1,"addresses":[{"type":"ble","peripheralAddress":""}]}
2025-04-19 13:55:13.153 DEBUG  BleScanner           Finishing waiter for query D:3840, resolving: true
2025-04-19 13:55:13.153 DEBUG  NobleBleClient       Stop BLE scanning for Matter Services ...
  noble scanStop +215ms
2025-04-19 13:55:13.153 INFO   ControllerDiscovery  Found 1 devices using identifier {"longDiscriminator":3840}
2025-04-19 13:55:13.153 DEBUG  ControllerDiscovery  Server addresses to try: ble://
2025-04-19 13:55:13.153 DEBUG  ControllerDiscovery  Try to communicate with ble:// ...
2025-04-19 13:55:13.153 INFO   PeerCommissioner     Establish PASE to device VP: 65521+32768
2025-04-19 13:55:13.154 DEBUG  BleChannel           Peripheral : Connect to Peripheral now (try 1)
2025-04-19 13:55:14.441 DEBUG  MdnsScanner          Sending 2 query records for 1 queries with 0 known answers. Re-Announce in 3 seconds
2025-04-19 13:55:14.563 DEBUG  BleChannel           Peripheral : Found services: fff6
2025-04-19 13:55:14.563 DEBUG  BleChannel           Peripheral : Handle service: fff6
2025-04-19 13:55:14.803 DEBUG  BleChannel           Peripheral : Handle characteristic: 18ee2ef5263d4559959f4f9c429f9d11 write
2025-04-19 13:55:14.804 DEBUG  BleChannel           Peripheral : Found C1 characteristic
2025-04-19 13:55:14.804 DEBUG  BleChannel           Peripheral : Handle characteristic: 18ee2ef5263d4559959f4f9c429f9d12 read indicate
2025-04-19 13:55:14.804 DEBUG  BleChannel           Peripheral : Found C2 characteristic
2025-04-19 13:55:14.805 DEBUG  BleChannel           Peripheral : Using MTU=0 bytes (Peripheral supports up to null bytes)
2025-04-19 13:55:14.806 DEBUG  BleChannel           Peripheral : Sending BTP handshake request: "656c040000000000ff"
2025-04-19 13:55:14.808 ERROR  BleChannel           Peripheral  disconnected while trying to connect, try again The specified UUID is not allowed for this operation.
```

on. two different devcies

stoprocent, 2025-04-19:
This is macOS issue nothing to do with `noble` - https://developer.apple.com/forums/thread/103557

stoprocent, 2025-04-19:
https://github.com/espressif/esp-matter/issues/637