stoprocent/noble #29: Cannot connect multiple devices simultaneously on Raspberry Pi

highflying, 2025-04-22:
Using a variant of `peripheral-explorer.ts`

-  I can connect to 2 devices at the same time on my Macbook (stays connected and stays running)
- Only connect to 1 device on my Raspberry Pi (and this is on an older kernel/OS that doesn't have the connecting issues in #12) and the discoverAsync for loop exits after handling the first device it connects to. 

```
const noble = require("@stoprocent/noble");

const addresses = [
  "e44510a1bd3e0ae1d364d263a964918e",
  "e050454ee7b5ce4ed12b436fbabbe133",
  "90842b5ee58a",
  "90842b4eab24",
];

const main = async () => {
  await noble.waitForPoweredOnAsync();
  console.log("noble started");

  for await (const peripheral of noble.discoverAsync()) {
    if (addresses.includes(peripheral.id) === false) {
      continue;
    }
    console.log("connecting to", peripheral.id);

    console.log("peripheral", peripheral.address || peripheral.id);
    console.log("discovered", peripheral.advertisement);

    peripheral.on("error", (err) => console.error("emitted error", err));

    console.log("status", peripheral.state);
    try {
      await peripheral.connectAsync();
      console.log("connected");
    } catch (err) {
      console.error("err connecting", err);
      continue;
    }

    try {
      console.log("status", peripheral.state);
      console.log("preservices");
      const { services } =
        await peripheral.discoverAllServicesAndCharacteristicsAsync();
      console.log("postservices");

      for (const service of services) {
        console.log("Service\x1b[34m", service.uuid, "\x1b[0m");
        for (const characteristic of service.characteristics) {
          console.log(
            "  | Characteristic\x1b[32m",
            characteristic.uuid,
            "\x1b[0m"
          );
          console.log("  |  | Type:", characteristic.type);
          console.log("  |  | Properties:", characteristic.properties);

          if (characteristic.properties.includes("read")) {
            let value = await characteristic.readAsync();
            console.log(`  |  | Value: 0x${value.toString("hex")}`);
          }

          const descriptors = await characteristic.discoverDescriptorsAsync();
          for (const descriptor of descriptors) {
            console.log(
              "  |  | Descriptor\x1b[33m",
              descriptor.uuid,
              "\x1b[0m"
            );
            const value = await descriptor.readValueAsync();
            console.log(`  |  |  | Value: 0x${value.toString("hex")}`);
          }
        }
      }
    } catch (err) {
      console.error("err getting services", err);
      continue;
    }
  }

  noble.stop();
  console.log("noble stopped");
};

void main();
```

[pi-output.txt](https://github.com/user-attachments/files/19855105/pi-output.txt)
[macbook-output.txt](https://github.com/user-attachments/files/19855104/macbook-output.txt)

stoprocent, 2025-04-22:
Sorry for closing now, but let's keep the track it in the other issue. I've already fixed it and the release will be done in a couple of minutes.

stoprocent, 2025-04-23:
I’ve reopened this issue because the case with the async generator loop is one where my latest fix does not work. This is a valid issue.

You can try adding `await startScanningAsync()` at the end of the for loop - if that makes it work, then I’m pretty sure I know what the fix needs to be.

stoprocent, 2025-04-24:
@highflying can you try the above?

highflying, 2025-04-25:
> [@highflying](https://github.com/highflying) can you try the above?

I still get the same, exits after interacting with the first hub.

Regina-v, 2025-07-02:
I'm running into the same issue. When I use `for await (const peripheral of noble.discoverAsync())`, it processes one peripheral and exits. When I use `noble.on('discover', cb)` or wrap the for await discoverAsync loop in a while loop like so:
```
while (true) {
  await noble.startScanningAsync([], true);
  console.log('Scanning for peripherals...');

  for await (const peripheral of noble.discoverAsync()) {
    await handleDiscovery(peripheral);
  }
  console.log('Scan complete, restarting...');
  await noble.stopScanningAsync();
}
```

then it does continue after the first peripheral. But after some time (sometimes after a few minutes, most often after 20-30 min of continuous scanning and connecting) I run into the issue that it takes more than 40 seconds to connect ( https://github.com/stoprocent/noble/issues/12 sounds familiar, I have also tested with version 2.1.7 and have the same issue there). 

And I'm not sure if this is related or a separate issue: if I add a timeout with 20 seconds to cancel the connection with `cancelConnect` then when this function is called I get the (error?) message "noble: unknown peripheral null connected!" which indicates that the cancelConnect somehow triggered the "being stuck" to change into "connected" but also canceled the connection??

I'm on a Raspberry Pi 4 Model B Rev 1.1 with Raspbian GNU/Linux 12 (bookworm) Linux 6.12.25+rpt-rpi-v8 and I would be more than happy if there anything I can do to help debugging!

We were using @abandonware/noble and tried to upgrade to node 22 which didn't work at all. That's why I'm currently trying to migrate to this library but also stuck.