stoprocent/noble #33: Cannot monitor Bluetooth status changes

pxgo, 2025-05-14:
Thanks to the author for his efforts, my electron app can operate Bluetooth.

During the test, I found that when the system Bluetooth was turned off and the following program was run, the Bluetooth status obtained was unsupported. When the system Bluetooth was turned on again, noble.on('stateChange') would not be triggered. The test code is as follows:

```javascript
const noble = require("@stoprocent/noble");

noble.on("stateChange", (state) => {
  console.log(`stateChange: ${state}`);
});

setInterval(() => {
  console.log(`timer: ${noble.state}`);
}, 2000);
```

I don't know if this is a bug. I have only tested it on the Windows 11 platform and don't know if other platforms have the same problem.

As of now, when users encounter the above scenario, they have to turn on the system Bluetooth first and then restart the electron app, which is troublesome.

stoprocent, 2025-05-15:
I will test this scenario and update the issue

pxgo, 2025-05-29:
> Hey I have some upcoming work on electron next week I will have a look then

Although the aforementioned test code can reproduce the issue in a pure Node.js environment, it would be preferable if testing could be conducted in an Electron environment.
 
Thanks!

closed 2025-07-12 by github-actions[bot]