stoprocent/noble #36: Getting a valid MTU value on webbluetooth

maelp, 2025-05-21:
The webbluetooth API doesn't seem to expose the MTU nicely, or am I missing something? What is the correct way to get the proper MTU value in those cases? Create a req/resp message with the device (if we control the device) to get it to send us the MTU it has negotiated? Or trying to write something big and catching an error (but which error?) and retrying with a smaller "guessed MTU" until it works?

maelp, 2025-05-21:
Just saw that you haven't yet ported the webbluetooth implem from the abandonware fork
https://github.com/abandonware/noble/blob/master/lib/webbluetooth/bindings.js

stoprocent, 2025-05-22:
I’ve removed `WebBluetooth` since it wasn’t being maintained, so it’s not really relevant to this issue. If you’re interested in maintaining it, we can definitely discuss bringing it back—but that would be better suited for a Pull Request, not a GitHub issue.

maelp, 2025-05-22:
Sure, let's discuss it here, and if it makes sense I could do a pull-request, although I think I'd rather do the one about Capacitor (which itself includes webbluetooth)

Apollon77, 2025-05-22:
When I remember correctly the MTU is not officially exposed by the WebBluetooth standard at all and is also only mentioned one time in the whole document https://webbluetoothcg.github.io/web-bluetooth/#exchange-mtu

So it really depends on the implementation of WebBluetooth how and if MTU is exposed ... and yes https://github.com/capacitor-community/bluetooth-le?tab=readme-ov-file#getmtu exactly states that tjsi is not available in the web :-)

stoprocent, 2025-05-23:
@maelp  I will close this one as we don't have a WebBluetooth driver. Im happy to review a PR, and we can continue the discussion there.