stoprocent/noble #45: Does this library support retrieving data from characteristics of type indicate?

LarsHulsmans, 2025-07-23:
So I'm trying to retrieve data from a ble device that has a characteristic of type indicate but if i use the provided example of subscribing to a notification it just doesn't get any data.  And if i try to use subscribeasync it just hangs.
Also i was wondering if there is any documentation on the options parameter in connect and connectAsync?
Kind regards.

stoprocent, 2025-07-23:
To be honest, I haven’t tested it in a while — I’ll give it a try when I get a chance.
As for your second question: yes, the params are defined in the `d.ts` file here:
[https://github.com/stoprocent/noble/blob/e2f1db9749265888b69517a3f7d88c974e32fdc3/index.d.ts#L22](https://github.com/stoprocent/noble/blob/e2f1db9749265888b69517a3f7d88c974e32fdc3/index.d.ts#L22)

However, I noticed a few days ago that the `connectAsync` interface currently doesn't accept any parameters, which is a bug on my end — I’ll need to fix that.

Apollon77, 2025-07-25:
yes indicate works. matter.js uses this